adafruit/Adafruit_BMP280_Library #5: Seems to get incorrect pressure on ESP8266

nicof38, 2016-05-23:
Using your library on ESP8266 12E, and while temperature reading seems to be ok, I'm getting a value for pressure that is always below 100 000 Pa (so lower than pressure I should get for an ambient one, at altitude close to 0).

As the algorithm is optimized versus the one available on bosch website(which is full 32 bits), Do you have some explanations about implementations ? or any check I can do on my side ?

Ihave already been using 2 others sensors and results are exactly the same.

arantius, 2016-07-21:
I just set this up on an ESP (via a NodeMCU), and thought I had the same problem (turns out because I was confused by units).  I'm now reading 101959.88 which converts to 30.109 inches of mercury, which is quite close to the 30.14 a weather website claims as the current value.

The code I see in this repository seems identical to that in the BMP280 datasheet.  Perhaps something else is happening for you?

aneek231, 2016-10-13:
I am also getting the pressure always below 100,000 Pa. Can you please tell me what is the problem ?

robertvandervoort, 2017-01-04:
Oddly, I had this problem earlier today for about an hour. Pressure went from 101468.6 pa (fluctuating) to 81414.78 stable (my tipoff something was wrong). At the 101 kpa reading it was showing I was about -12m altitude and when it hits 81 kpa said +1800! Good thing I wasn't using the sensor to FLY anything right?!!

I am also using nodeMCU programmed through Arduino IDE and this library. Nothing has changed code wise, and the library remains the same (just began doing this yestarday so library code is unchanged). I'm real curious to see if anyone figures out what is going on.

Anecdotally, last night I was getting a fairly accurate reading for altitude, around 26m (I am in Houston, TX). And the pressure was about the same (from memory).

robertvandervoort, 2017-01-04:
last comment.. have the altitude right on by using the right value for the parameter, but I'd suggest a future improvement for these sensors to add a reset function as outlined in the bmp280 datasheet...

> 4.3.2 Register 0xE0 “reset”
> The “reset” register contains the soft reset word reset[7:0]. If the value 0xB6 is written to the
> register, the device is reset using the complete power-on-reset procedure. Writing other values
> than 0xB6 has no effect. The readout value is always 0x00.

robertvandervoort, 2017-01-06:
ok I lied :)

I am logging every one minute and dashboarding said results.. Part of my project is an ADC (the 1115 chip from adafruit) and I see when the analog lines with no pull-down (and nothing attached) get noisy, I assume from USB noise getting through the ESP-8266 and into the 3v3 vin of the 1115, after a few minutes or less than an hour is when the BMP loses its mind... Oddly, power cycling it doesn't seem to help, I have to power down the entire unit (maybe power is getting in from the i2c pins?) and it comes back to life. If I run the whole unit on battery only, there's harly any noise on the 1115... The BMP seems to do great.. UNTIL... I had it outside and it is like 40 degrees out today.. and it just wouldn't work at all.. I rebooted and it went into a loop because it is waiting on the BMP to initialize.. i removed the loop code to just power past it and when I poll the sensor, all I get is this:

Temperature = 32.00 *F
Pressure = 0.00 Pa
Approx altitude = 145446.73 ft

yeah I'm converting to F and ft because I'm a wuss like that... 

Should I assume this sensor is kaput?

ronnicek, 2017-07-28:
Also getting under 1000hPa.. but I have a little meteo station, so I found out that when I calculate pressure from sensor + 40 then I get same result as on meteo station.

DavidRTucker, 2018-06-10:
What overall value of pullup resistance?  Perhaps it is worth checking the waveform on the I2C bus?  Also try reading the sensor with SPI instead and see if that has the same issue?

andzer, 2018-07-29:
I believe that you have to calculate sea level pressure. After this, it's accurate.

DavidRTucker, 2018-07-29:
I have tried reading six sensors together to improve the accuracy two on I2C and four on SPI. Obviously this is an improvement but not stable enough to use for differential altitude with a tolerance of ±100 mm.  I am building a couple of twelve channel sensing systems but I am not holding out much hope for any great improvements or achieving my objective with the BMP280.

andzer, 2018-07-29:
Here is the calculation formula to find sea level pressure.

andzer, 2018-07-29:
https://keisan.casio.com/exec/system/1224575267

ladyada, 2019-12-29:
closing due to lack of activity

closed 2019-12-29 by ladyada