adafruit/Adafruit_BMP280_Library #18: Interference on the I2C data bus

fu-hsi, 2018-12-08:
- Arduino board: **Wemos D1 Esp-Wroom-02**
- Arduino IDE version (found in Arduino -> About Arduino menu):  **1.8.5**

BMP280 not works correctly with SHD31-D in parallel after upgrade library version to 1.0.2.
Without SHD31-D problem still exists.
Problem occur after one good reading or more.

| Version | Temperature | Pressure / 100 | Note
|--------| ------------- |----------|----------------------|
| 1.0.2    |  -99.99          |155.69    |                                |   
| 1.0.1    | -99.99           |1278.14  |                                |
| 1.0.0    | OK                | OK          | OK but not for long |

After disconnect BMP280, SHT starts working properly (from id 142689).

```
    id                   dt  sht_temp  sht_hum  bmp280_press  bmp280_temp  
------  -------------------  --------  -------  ------------  -------------
142698  2018-12-08 21:27:56     23.31    52.02       -155.69          99.99
142697  2018-12-08 21:26:41     23.34    51.96       -155.69          99.99
142696  2018-12-08 21:25:26     23.34    51.71       -155.69          99.99
142695  2018-12-08 21:24:11     23.38    52.29       -155.69          99.99
142694  2018-12-08 21:22:56     23.34    51.89       -155.69          99.99
142693  2018-12-08 21:21:41     23.38    51.78       -155.69          99.99
142692  2018-12-08 21:20:26     23.41    51.81       -155.69          99.99
142691  2018-12-08 21:19:11     23.37    51.89       -155.69          99.99
142690  2018-12-08 21:17:56     23.40    52.54       -155.69          99.99
142689  2018-12-08 21:16:41     23.42    51.73       -155.69          99.99
142688  2018-12-08 21:15:26    (NULL)   (NULL)       -155.69          99.99
142687  2018-12-08 21:14:11    (NULL)   (NULL)       -113.46          99.99
142686  2018-12-08 21:12:56     23.28    52.02        983.75          24.73
142685  2018-12-08 21:11:41     23.28    52.52        983.83          24.69
142684  2018-12-08 21:10:26     23.28    52.81        983.80          24.67
142683  2018-12-08 21:09:11     23.23    52.57        983.75          24.65
142682  2018-12-08 21:07:56     23.24    52.34        983.95          24.57
142681  2018-12-08 21:06:41     23.24    52.31        983.88          24.65
142680  2018-12-08 21:05:26     23.24    52.73        983.90          24.68
142679  2018-12-08 21:04:11     23.27    52.47        983.82          24.69
142678  2018-12-08 21:02:56     23.21    52.44        983.84          24.63
142677  2018-12-08 21:01:41     23.20    52.62        983.93          24.56
142676  2018-12-08 21:00:26     23.21    52.42        983.91          24.53
142675  2018-12-08 20:59:11     23.21    52.42        983.95          24.54
142674  2018-12-08 20:57:56     23.23    52.87        984.01          24.56
142673  2018-12-08 20:56:41     23.21    52.27        984.00          24.61
142672  2018-12-08 20:55:26     23.20    52.21        984.18          24.62
142671  2018-12-08 20:54:11     23.18    52.10        984.22          24.73
142670  2018-12-08 20:52:56     23.21    52.33        984.56          24.85
142669  2018-12-08 20:51:41     23.24    52.12        946.74          25.18
142659  2018-12-08 20:38:04    (NULL)   (NULL)       -155.69          99.99
142658  2018-12-08 20:36:49    (NULL)   (NULL)       -155.69          99.99
142657  2018-12-08 20:35:34    (NULL)   (NULL)       -106.75          99.99
142656  2018-12-08 20:34:14    (NULL)   (NULL)       -113.21          99.99
142655  2018-12-08 20:32:59    (NULL)   (NULL)        948.90          23.80
142654  2018-12-08 20:32:03    (NULL)   (NULL)       1278.14         -99.99
142653  2018-12-08 20:30:48    (NULL)   (NULL)       1278.14         -99.99
142652  2018-12-08 20:29:33    (NULL)   (NULL)       1278.14         -99.99
142651  2018-12-08 20:28:18    (NULL)   (NULL)       1278.14         -99.99
142650  2018-12-08 20:27:03    (NULL)   (NULL)       1278.14         -99.99
142649  2018-12-08 20:25:48    (NULL)   (NULL)        984.34          23.34
142648  2018-12-08 20:24:05    (NULL)   (NULL)       1278.14         -99.99
142647  2018-12-08 20:21:20    (NULL)   (NULL)       1278.14         -99.99
```
Initialization:
```cpp
#include <Adafruit_SHT31.h>
#include <Adafruit_BMP280.h>

Adafruit_SHT31 sht;
Adafruit_BMP280 bmp;

void setup(void) {
    Wire.begin(2, 14); // D4, D5
    sht.begin(0x44);
    bmp.begin(0x76);
}
```
Maybe it's just a sensor fault. I replaced it into BME280 and I think it works.
If it stops, I'll write.

sslupsky, 2019-05-01:
I noticed that the BMP280 library uses a reference for the wire interface so the I2C bus can be specified:

https://github.com/adafruit/Adafruit_BMP280_Library/blob/9912b7f26686c28792ef46a741e65e503b66d883/Adafruit_BMP280.cpp#L76

whereas the SHT31 library is hard coded to use the default I2C bus:

https://github.com/adafruit/Adafruit_SHT31/blob/4626ddd490e865a9916967eb1c881b8a44637682/Adafruit_SHT31.cpp#L25

Is it possible that the two libraries are conflicting with each other and causing the data corruption you see?

hoffmannjan, 2019-05-29:
fixed in #23

closed 2019-05-29 by hoffmannjan

sslupsky, 2019-05-29:
@hoffmannjan FYI, I made a similar change and PR to the Adafruit SHT31 library.

https://github.com/adafruit/Adafruit_SHT31/pull/16