adafruit/Adafruit_BMP280_Library #21: Exception on ESP8266 with current Arduino library

borisneubert, 2019-04-19:
On  ESP8266 platform with the latest Arduino library, the ESP8266 immediately generates an exception when an object of class Adafruit_BMP280 is instantiated. Reverting to commit #7dca355 works just fine. 

Please let me know what additional information you need to reproduce or eliminate the issue.

ladyada, 2019-04-19:
yes please provide a lot more info - what boards, what versions, error codes

borisneubert, 2019-04-19:
Thank you for the fast reaction!

Board is d1_mini. 

`PlatformIO 3.6.6 is currently the newest version available.`

```
Updating tool-scons                      @ 2.20501.7      [Up-to-date]
Updating tool-unity                      @ 1.20403.0      [Up-to-date]
Updating contrib-pysite                  @ 2.27.190418    [Up-to-date]
Updating contrib-piohome                 @ 2.0.0          [Up-to-date]
Updating tool-pioplus                    @ 2.1.3          [Up-to-date]

Platform Manager
================
Platform Espressif 8266
--------
Updating espressif8266                   @ 2.0.4          [Up-to-date]
Updating tool-mkspiffs                   @ 1.200.0        [Up-to-date]
Updating tool-esptool                    @ 1.413.0        [Up-to-date]
Updating tool-espotapy                   @ 1.1.3          [Up-to-date]
Updating framework-arduinoespressif8266  @ 2.20500.190223 [Up-to-date]
Updating toolchain-xtensa                @ 1.40802.0      [Up-to-date]
```

Serial console output of the programm with current version of Arduino_BMP280 is
```

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
￼
 
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v951aeffa
~ld
```

(repeating forever)

The program runs normally if I remove the declaration of the Arduino_BMP280 object.

It also runs normally with commit #7dca355 of the library. Did some bisection:

#2868f39 fine
#7e9dd6c exception
#3805bc5 fine
#ecc4de2 exception
#8ae0d7b fine
#4ff54f8 does not compile, error:
```
.piolibdeps/Adafruit BMP280 Library/Adafruit_BMP280.cpp: In member function 'bool Adafruit_BMP280::begin(uint8_t, uint8_t)':
.piolibdeps/Adafruit BMP280 Library/Adafruit_BMP280.cpp:72:14: error: 'a' was not declared in this scope
_i2caddr = a;
^
*** [.pioenvs/d1_mini/libc11/Adafruit BMP280 Library/Adafruit_BMP280.cpp.o] Error 1
```
#d60796f does not compile with same error
#2daef4a does not compile with same error

ladyada, 2019-04-19:
are you using adafruit products?

borisneubert, 2019-04-19:
Issue does occur even with no device at I2C bus at all.

ladyada, 2019-04-19:
ok - are you using any adafruit products?

borisneubert, 2019-04-19:
I have tons of modules lying around from various sources. How can I tell if it is from Adafruit?

ladyada, 2019-04-19:
it has the name Adafruit and logo on it

borisneubert, 2019-04-19:
The BMP280 on my desk actually has not. This issue is not related to hardware. Guess the current version of the library will not work with any module.

ladyada, 2019-04-19:
OK please ask the vendor you bought the hardware from for a first round of tech support - they have engineers on staff that can help  you :)

borisneubert, 2019-04-19:
Well, actually I wanted to help with the software and fix the issue that is likely to affect any user of the library no matter what the source of the hardware is and so contribute to the efforts you put in providing the library in the first place. Asking the vendor for tech support is meaningless because it is not a hardware issue. That's why I came here. Sorry that you take that position.

Well I solved the issue for myself.

closed 2019-04-19 by borisneubert