adafruit/Adafruit_BMP280_Library #22: Latest commit (7e9dd6c) doesn't work with Arduino Mega 2560

KeltenP, 2019-04-22:
What platforms has the latest commit been tested on? Using the code from the latest commit (7e9dd6c) causes my program that uses the BMP280 to completely stop working on the Mega 2560. When calling the begin() function my board just power cycles. Reverting to commit 3805bc5 makes it work again.

I can post more information if required.

hoffmannjan, 2019-04-22:
There was a typo in i2c assigment, I tested It today with ESP32 & UNO and all works fine, using i2c as well as SPI.
Sorry for the problems, please confirm that `master` now works for you.

KeltenP, 2019-05-27:
Just got around to testing this out again, everything seems to be working now. Thanks for the fix!

closed 2019-05-27 by KeltenP