adafruit/Adafruit_BMP280_Library #24: Compile fails if SPI_INTERFACES_COUNT = 0

sslupsky, 2019-05-06:
- Arduino board:  Custom SAMD board

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.9

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **LIST REPRO STEPS BELOW**

I am using a custom SAMD board and configured the board to use I2C only and there are no SPI ports configured.  That is, SPI_INTERFACES_COUNT = 0.

If SPI_INTERFACES_COUNT = 0 in variants.h, the compilation fails with the following error message:

/Arduino/libraries/Adafruit_BMP280_Library/Adafruit_BMP280.h:156:53: error: 'SPI' was not declared in this scope
   Adafruit_BMP280(int8_t cspin, SPIClass *theSPI = &SPI);

I think each of the code sections for SPI and I2C need to be enclosed with a conditional compiler directive.  For example:

#if SPI_INTERFACES_COUNT > 0
blah blah
#endif

ladyada, 2019-05-06:
yep it would have to - want to try submitting a PR for these changes? you can test with your custom samd board then :)

sslupsky, 2019-05-06:
I'll work on something and send it to you.  I encountered this problem with another library as well.  I used the conditional compiler directive approach to resolve the issue.  Another approach discussed with the maintainer was to forward declare the SPI and Wire objects.  His concern was whether or not SPI_INTERFACES_COUNT and WIRE_INTERFACES_COUNT were used with other cores.  I think he may have meant architectures.

Any thoughts on either approach?

sslupsky, 2019-05-06:
While working on the this issue I came across the following line:

https://github.com/adafruit/Adafruit_BMP280_Library/blob/9912b7f26686c28792ef46a741e65e503b66d883/Adafruit_BMP280.cpp#L133

I think this line should be:

return _spi->transfer(x);

ladyada, 2019-05-06:
you're right! wanna fix that too? :D

ladyada, 2019-05-06:
i think SPI_INTERFACES_COUNT is safe to use, its in modern variant files like https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/variants/feather_nrf52840_express/variant.h#L106

sslupsky, 2019-05-06:
Yeah, I fixed that up.  

This particular library is written to share a some methods across SPI and I2C.  So I think the forward declaration is the way to go.  I'll do a little more testing and then send a PR your way.

ladyada, 2019-05-06:
sweeet!

hoffmannjan, 2019-05-29:
merged in #23

closed 2019-05-29 by hoffmannjan