adafruit/Adafruit_BMP280_Library #27: Enabling "takeForcedMeasurement()"?

Jilocasin, 2019-06-05:
Right now when setting the BMP280 in FORCED mode at startup once, I experience reading out the same value for temperature and pressure over and over. This is probably due to the sensor not updating the readout values periodically (like in NORMAL mode), but rather waiting for a trigger to do so.

Compared to the BME280 library the "takeForcedMeasurement" method is still commened out with a "todo" annortation. As the same method in the BME280 library exists and has the exact same content, I wanted to ask if the method could be safely enabled in this library as well to allow using the FORCED mode with the BMP sensor at all.

tcpipchip, 2019-10-02:
i am having the same problem :(

Jilocasin, 2019-10-02:
After enabling the commented part in the BMP library, I was able to force single measurements, just like the BME library is capable of. You can give that a try.

tcpipchip, 2019-10-02:
Now it´s working very fine!

ladyada, 2019-10-02:
hiya can you tell us exactly what line?

Jilocasin, 2019-10-02:
I was talking about the following:
https://github.com/adafruit/Adafruit_BMP280_Library/blob/master/Adafruit_BMP280.h#L166
https://github.com/adafruit/Adafruit_BMP280_Library/blob/master/Adafruit_BMP280.cpp#L399

Sorry if this isn't the appropriate way to "cite" content in a repo. Still new to github

ladyada, 2019-10-02:
oh yeah it isnt tested, so no guarantee it works!

R3Spected, 2019-10-10:
Had to add
`BMP280_REGISTER_STATUS = 0xF3,`
aswell.

tcpipchip, 2019-10-10:
nice

WX-PLU, 2020-08-02:
My temp and pressures reads zero. R3Spected, what file and line did you add that?

tcpipchip, 2020-08-02:
https://zephy-platformio.blogspot.com/2020/03/nina-b112-zephyr-platformio.html?m=1

denisandroid, 2020-08-30:
I confirm, `takeForcedMeasurement ()` works on BMP280 via I2C(esp8266).

But sometimes, rarely, once there was a value of 0 at startup, but it is not accurate because of this or not.

closed 2022-01-11 by caternuson