adafruit/Adafruit_BMP280_Library #35: Pa or hPa?

bleckers, 2019-11-13:
In the library cpp file, the code states:

```
/*!
 * Reads the barometric pressure from the device.
 * @return Barometric pressure in hPa.
 */
float Adafruit_BMP280::readPressure() {
```

However in the example it states:

```
    Serial.print(F("Pressure = "));
    Serial.print(bmp.readPressure());
    Serial.println(" Pa");
```

Are these libraries returning Pa or hPa?

bleckers, 2019-11-13:
Getting values of 101631.3 leads me to believe this library returns Pa, unless we are on Venus or something?

ladyada, 2019-11-13:
yes it returns Pa, if you submit a PR for the doc fix, we'll merge!

ladyada, 2019-11-14:
you have to PR it next

kevinjwalters, 2020-03-07:
This is _inconsistent_ with the specification/guidelines on https://learn.adafruit.com/using-the-adafruit-unified-sensor-driver/how-does-it-work#standardised-si-units-for-sensor-data-2-2 which states hPa (hecto-Pascals). I noticed this on the video for CLUE: https://www.youtube.com/watch?v=Nz38vF2W8BI&t=11s (look closely at values on y axis)

ladyada, 2020-03-07:
readPressure() is Pa, but the Unified sensor example (sensortest) is in hPa