adafruit/Adafruit_BMP280_Library #37: Typo correction

kostashellas, 2020-01-14:
Hello team,
the below alternative I2C address does not work, I guess because of a typo error  **\**

#define BMP280_ADDRESS_ALT                                                     **\**
  (0x76)                     /**< Alternative I2C address for the sensor. */

ladyada, 2020-01-14:
ok! would you be able to submit a tested PR? then we'd merge it  :)

caternuson, 2021-01-20:
Just took a quick look at this. The `\` is being added by `clang-format` to break up the "long" line. It's not a typo.

Your example above is all on one line while the actual library code is on two lines:
https://github.com/adafruit/Adafruit_BMP280_Library/blob/dd4b67f9bdbea7450f09007be70951fa8d78ac2b/Adafruit_BMP280.h#L33-L34
It needs to be on two lines. Specifically - the continuation character needs to be the last character on the line.

Can you clarify how it does not work?

If you are getting a compile error, see if something has altered your local copy of `Adafruit_BMP280.h` for some reason.

If you can't get the sensor to work with that address, make sure you have the address select jumpered correctly on the breakout.

This example compiles are runs as expected:
```cpp
#include <Adafruit_BMP280.h>

void setup() {
  Serial.begin(9600);
  while (!Serial);

  Serial.println(BMP280_ADDRESS);
  Serial.println(BMP280_ADDRESS_ALT);
}

void loop() {
}
```

[image]

caternuson, 2021-07-31:
Closing due to lack of response.

closed 2021-07-31 by caternuson