adafruit/Adafruit_BMP280_Library #47: Min Value swapped with Max Value

AlexZibin, 2021-01-20:
I've compiled the example code, everything woks great except that **Min Value** is swapped with **Max Value**.
Here's the output: 
```
BMP280 Sensor event test
------------------------------------
Sensor:       BMP280
Type:         Ambient Temp (C)
Driver Ver:   1
Unique ID:    280
Min Value:    85.00
Max Value:    -40.00
Resolution:   0.01
------------------------------------

Temperature = 23.46 *C
Pressure = 1012.14 hPa
```

ladyada, 2021-01-20:
woops, can you submit a PR to fix / swap the numbers? :)

caternuson, 2021-01-20:
See #48 . Looks like same typo was in the pressure characterization also. Fixes both.

caternuson, 2021-01-20:
Closing. Fixed by #48. Thanks again @AlexZibin for pointing this out.

closed 2021-01-20 by caternuson