adafruit/Adafruit_BMP280_Library #56: different chip ID

WRFleete, 2021-08-15:
I likely had a slightly different board to the one pictured but I found I could only get it to work with the following snippet, either I have a slightly different device with different chip ID or the code is a bit outdated, my board had the 0x76 I2C address and a 0x60 chip ID.

```
if (!bmp.begin(BMP280_ADDRESS_ALT, 0x60)) {
  //if (!bmp.begin()) {
    Serial.println(F("Could not find a valid BMP280 sensor, check wiring or "
                      "try a different address!"));
    while (1) delay(10);
  }
```

had to use the I2C scanner and a bit of additional code and reading the datasheet to figure out where the chip ID register was and what the value was  (snippet below)
```
void loop() {
  int nDevices = 0;

  Serial.println("Scanning...");

  for (byte address = 1; address < 127; ++address) {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    Wire.beginTransmission(address);
    byte error = Wire.endTransmission();

    if (error == 0) {
      Serial.print("I2C device found at address 0x");
      if (address < 16) {
        Serial.print("0");
      }
      Serial.print(address, HEX);
      Serial.println("  !");
        Wire.beginTransmission(address);
        Wire.write(0xD0);
        Wire.endTransmission();
  Wire.requestFrom(address, 1);
        Serial.print("Chip ID: ");
        Serial.println(Wire.read(),HEX);
      ++nDevices;
    } else if (error == 4) {
      Serial.print("Unknown error at address 0x");
      if (address < 16) {
        Serial.print("0");
      }
      Serial.println(address, HEX);
      
    }
  }

```

closed 2021-08-15 by ladyada

WRFleete, 2021-08-16:
No explanation for closing issue? I want to know.

ladyada, 2021-08-16:
as you say, you likely had some other chip