adafruit/Adafruit_BMP280_Library #59: include I2C bitbang

hydrastarmaster, 2021-08-29:
Projects sometimes get very crowded. I have a mandatory :) touchscreen that overlaps both the BMP280 I2C addresses. The solution would be (1) to place a physical multiplexer (but it's quite nonsense, being a SW solution possible) or (2) to use a couple of ports, among the many still free, to bitbang the I2C comms. In the repository, e.g. the BitBang_I2C could be integrated into Adafruit BMP280 to meet this need.

ladyada, 2021-08-29:
we'd accept a bitbang i2c implementation in Adafruit BusIO so its abstracted out, then it could be added easily to any sensor. please submit a PR when you can!

hydrastarmaster, 2021-08-29:
Thanks. Message delivered to bitbank2.
https://github.com/bitbank2/BitBang_I2C/issues/27