adafruit/Adafruit_BMP280_Library #64: Failing Gracefully?

sellensr, 2021-11-25:
- Arduino board:  Itsy Bitsy M0 Express and others

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.16 and others

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): 

begin() does not return if the sensor is mis-wired, e.g. power lead is not connected. I would have hoped for a timeout after a while and returning with a failure. I suspect this may be farther downstream in the unified sensor or bus io code.

Functions like readTemperature() don't return if called before begin()

None of this is really unexpected, but my students are finding ingenious ways to do things in the wrong order and more resilient software might set a good example. A start at a simple solution might have begin() set a flag that a sensor has already been found and have the others test the flag before trying to talk to the sensor. I'm wondering if this is a problem worth solving...

caternuson, 2021-11-25:
The `begin()` function attempts to do some checking and return true/false as appropriate. However, for things like an unpowered sensor, it may not be possible, or at least gets tricky. This library sort of assumes the I2C bus is OK. You're correct that it would be up to the lower level libraries to attempt to detect these kinds of things.

The general way to check things before proceeding is shown in the example:
https://github.com/adafruit/Adafruit_BMP280_Library/blob/9485eabfe894482e7302eb2697d8f7e41f0d38b5/examples/bmp280test/bmp280test.ino#L36-L40

If the call to `begin()` is not returning, then there's likely not much this library can do to remedy that, since it's probably one of the calls to the lower libraries that is actually hanging.

sellensr, 2021-11-26:
The equivalent areas of the BME280 library code set the _sensorID variable and print some additional diagnostic information in the test example code. That would be easy to implement here and would pave the way for testing _sensorID to see if a valid sensor has been found and initialized before trying to read from it. _sensorID was declared, but not initialized or set in the current code.

sellensr, 2021-11-26:
https://github.com/adafruit/Adafruit_BME280_Library/blob/a9c82197bc8a238f205196ed83e0b8efcfa6f171/examples/bme280test/bme280test.ino#L42-L56

caternuson, 2021-11-26:
That's essentially the same thing. Using the return of the call to `begin()`. It just prints more information when it fails.

Checking chip ID is one of the things done in `begin()`:
https://github.com/adafruit/Adafruit_BMP280_Library/blob/9485eabfe894482e7302eb2697d8f7e41f0d38b5/Adafruit_BMP280.cpp#L99-L100

But successfully reading that requires the device has power and is wired correctly, etc.

sellensr, 2021-12-02:
I think that's as fixed as it can get locally -- anything else would need to happen in the unified sensor library or wire

closed 2021-12-02 by sellensr