adafruit/Adafruit_BMP280_Library #81: Library is missing dependency list (e.g., Adafruit_I2C)

DeflateAwning, 2023-06-18:
When using this library in Platform.io, it doesn't install the required dependencies correctly. This is presumably an issue in other development environments as well. 

Please set the dependency list correctly and fully. Thanks!

WackoKacko, 2023-06-21:
Was about to post the same issue. Library doesn't work on PlatformIO

DeflateAwning, 2023-06-21:
For me, adding the Adafruit BusIO library to the lib_deps made it work, but that shouldn't be required

WackoKacko, 2023-06-21:
Happy it worked for you. Just tried the same and it still failed to build for me. Any further tips?

DeflateAwning, 2023-06-21:
What's your error? Open a separate issue with your specific error

ladyada, 2025-12-10:
full dependancies are here: https://github.com/adafruit/Adafruit_BMP280_Library/blob/master/library.properties#L10 and they are recursive, which Arduino CLI handles fine (?)

closed 2025-12-10 by ladyada